ExpressLRS/ExpressLRS #3258: ExpressLRS partial desync and inability to reestablish connection

PauliusMorku, 2025-06-21:
## Current Behavior
ExpressLRS link enters a partial desync state where Link Quality (LQ) drops to very low values with most packets being lost, but the connection doesn't fully disconnect. The link does not recover on its own and requires a manual reset of either TX or RX to restore normal operation. This appears to happen because occasional packets still get through, preventing ExpressLRS from entering scanning mode that would allow proper re-synchronization.

## Steps to Reproduce
1. Set up Happymodel EP1 Dual receiver as TX connected to DTQSYS Head Tracker running HeadTracker firmware
2. Set up second Happymodel EP1 Dual receiver as RX
3. Establish normal ExpressLRS link with good LQ
4. Connect to head tracker via serial and perform settings save operation to flash. **Note:**: Issue can occur randomly during normal operation, but saving head tracker settings is a reliable reproduction method.
5. Observe LQ drops dramatically and stays low
6. Link remains "connected" but with very poor packet success rate
7. Only manual TX/RX reset restores normal operation (reseting headtracker is not necessary meaning it continues to send valid packets after the issue occurs).

## Possible Solution (based on my very limited understanding)
The issue may be related to how ExpressLRS handles irregular telemetry timing or gaps in Crossfire data. Since the head tracker and ExpressLRS module are separate devices, flash operations on the head tracker should not affect radio synchronization. Possible areas to investigate:

- Possibly some invalid operation is performed by the headtracker but I think ExpressLRS should be immune to this
- Crossfire protocol timing handling in ExpressLRS firmware
- Recovery mechanisms when partial desync occurs vs full disconnect

## Details
This severely impacts head tracker usability as the link becomes unreliable and requires manual resets. The head tracker and ExpressLRS module are separate devices communicating via Crossfire protocol, so head tracker operations (like flash writes) should not affect radio synchronization.
The issue suggests a problem with ExpressLRS recovery mechanisms - when synchronization is partially lost but some packets still get through, the system doesn't recognize it needs to re-establish sync.

## Environment

- TX hardware: Happymodel EP1 Dual receiver (2.4GHz) flashed as TX
- RX hardware: Happymodel EP1 Dual receiver (2.4GHz)
- Handset model: DTQSYS Head Tracker running https://github.com/headtracker/HeadTracker/releases/tag/v2.21
- OpenTX version: Not applicable
- ExpressLRS version: Tested on both 3.5.5 and 3.5.6 (TX & RX matching versions)
- Packet Rate: all tested modes behave the same (tested most of them, in fact I tried most of the settings that can be changed with lua script)
- Telemetry Ratio: telemetry ratio did not seem to impact this issue
- user_defines: none

CapnBry, 2025-06-22:
There definitely can be issues like this where the RX stops receiving, but not long enough for it to go to "disconnected" state and then gets in this quasi-connected state where it randomly is on the correct channel enough for the connection to stay alive, but with very low LQ.

The headtracker's code is likely stopping the transmitter for long enough for it to fall into that golden zone. The TX will keep transmitting for a brief time if the handset (in this case the head tracker?) stops sending channels, but the duration is pretty short. Seems like the HT author could fix their side so it either is faster or slower to avoid this issue and not trigger this architectural flaw in our OTA protocol.

CapnBry, 2025-06-23:
Well we'll disagree that we should just keep sending data after the handset stops providing sticks information. Keeping stale channels info coming prevents the other side from ever initiating any failsafe procedures, which it needs to do for safe operation.

Regarding the OTA, yes there's not enough bandwidth to include sync data in every packet so the sync packet is its own type of packet, which takes the place of channels info. Because it replaces the channels info, it isn't sent very frequently. The odds of a receiver that is out of sync in this scenario getting that sync info is relatively low so it never picks back up in the proper place again. 

One of the other devs and I have been discussing this sort of issue for quite a while and possible solutions in our OTA but your solution would be to get the headtracking software to keep sending channels so the link stays intact, or shut down the TX while they would be paused to allow the RX to completely disconnect and resync. It really should not be stopping the link if you're flying though IMO. This isn't a problem for handsets so it is something the headtracking software can improve.

CapnBry, 2025-06-25:
I've been thinking more about this and checked our source. The "No handset" timeout and the stale channels timeout are 1000ms, which seems longer than something should really take to start sending channels again. Is the headtracker powering off the TX then powering it back on when the config is saved? Something stopping sending for a full second seems unusual.