ExpressLRS/ExpressLRS #3263: Trivially Prevent DFU Hijacking

amstan, 2025-06-28:
## Current Behavior

I have an annoying case of [dfu hijacking](https://betaflight.com/docs/wiki/guides/current/dfu-hijacking) caused by the ELRS module. This is happens to be on a [flywoo goku-f405-hd-1-2s-elrs-aio-v2 (aka FLYWOOF405HD)](https://flywoo.net/products/goku-f405-hd-1-2s-elrs-aio-v2), but (since apparently this company name is evil or something) I believe this would happen with plenty of other controllers (eg: a speedybee with ELRS on a specific port and not a lot of access into the craft to disconnect the FC-RX wires).

## Explanation
This happens becase the [stm32 bootloader](https://www.st.com/resource/en/application_note/an2606-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf) first checks for UART traffic (looks particularly for 0x7F bytes), if it sees any it doesn't even bother checking USB DFU anymore and just gets stuck in a UART bootloader mode (which won't be productive since ELRS is not going to be programming the FC any time soon). It's possible the 0x7F byte gets sent often as part of ELRS communication payloads (I'm not an expert here), or.... it might something as silly as a different baud rate garbage getting interpreted as 0x7F.

## Steps to Reproduce

1. Have board with a FC and ELRS hardwired together on an unfortunate FC UART port (read: 50% chance you have this).
2. Try to enter bootloader for just the FC (hold the FC boot0 and plug usb, or.... just `dfu` on inav CLI)
3. Sadness: Do not see any bootloader or any usb device enumerate on the host (checked with `dmesg -w`)

## Workaround

* Hold bootloader for ELRS at the same time as the first button while pluging usb. This is quite annoying as it requires even more finger gymnastics, sometimes with tools (since the buttons are small and in hard to reach places).

## Possible Solution

My recommendation is 100ms delay on ERLS booting before it muxes the UART port. Or perhaps before it sends the first packet. I believe this might be enough to skip the 0x7F check leading into the UART dead end in the STM32F403 bootloader (see [AN2606](https://www.st.com/resource/en/application_note/an2606-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf), page 140). Once the FC detected the usb cable (which should happen fairly quickly), all UART devices are free to say anything (I think 100ms should be plenty!).

amstan, 2025-06-28:
I'm happy to take a stab at a PR for this, but I'm not near any oscilloscopes to prove my theories and iterate on firmware for this.

pkendall64, 2025-07-05:
The flywoo has an ESP8285 based RX, these chips always output a message on the UART when they boot. It is the bootloader ROM that outputs the message and the ELRS firmware has no control over this.

closed 2025-07-07 by pkendall64