ExpressLRS/ExpressLRS #3271: Helloradiosky HR8E only provides Voltage from Ext Bat Port not Voltage on PWM Pins

mama99de, 2025-07-04:
## Current Behavior

Helloradiosky HR8E only provides Voltage from Ext Bat Port not Voltage on PWM Pins

## Steps to Reproduce

Connect Power to PWM Servo Pins, RxBt always shows 0V

RxBt only shows Voltage if Ext-Bat is connected, not also for Servo Pins(if connected only), like descriped an RadioMaster PWN RX is doing it.

## Possible Solution (Not obligatory)
None

## Details
None

## Your Environment

- TX hardware: RM GX16/HRS16/RM Pocket
- RX hardware: Helloradiosky HR8E
- Handset model:-
- ETX version : 2.11.2
- ExpressLRS version (TX & RX MUST MATCH): 3.5.6 LBT
- Packet Rate: 100
- Telemetry Ratio:1:8
- user_defines: EU/LBT

mama99de, 2025-07-04:
To be honest, never tested it with the stock HRS Firmware, always updating my RX/TX to the latest version

mama99de, 2025-07-04:
According to HK, it should work:
https://hobbyking.com/en_us/helloradiosky-hr8e-elrs-9-channel-2-4ghz-receiver-black.html
https://cdn-global-hk.hobbyking.com/media/catalog/product/h/o/hobbyking-hr8e_8_.jpg
Got mine from Derkum GER/EU

mha1, 2025-07-05:
In contrast to RM ER receivers HRS HR7x/HR8x receivers can only measure external voltages. As a workaround you can externally connect one of the PWM VCC pins (middle pin) to the EXT-BAT + pin.