ExpressLRS/ExpressLRS #3299: Airport mode: the TX won't connect to the RX if the RX starts and activates wifi mode before TX is online.

Bwooce, 2025-08-09:
## Current Behavior

Using a Bandit 915 TX and and BR1 RX, both in Airport mode, both on v3.5.6 firmware. The TX is running "naked" and connected to my computer, not in a controller at all.  RX to a flightcontroller (ESP32 Ardupilot, but a new board and I've been working hard on that code).

Both running at 

* If the TX is started before the 60s wifi timeout then the TX and RX will bind and chat happily. 
* If the TX is stated after the 60s wifi timeout then the TX never connects
* [possibly not the same issue] If I press the bind button on the TX then it sends the bind packets on the USB connection (I captured and decoded them, because they didn't look like MAVlink). I don't know if it's sending them over the radio...but it doesn't work.
* If I power-cycle the RX then the TX will connect to it
* If the RX and TX connect before the 60s timeout then the TX will reconnect automatically, even after the 60s re-timeout (I guess)
* If the RX and TX are not connected then the RX wifi will start, but they'll never bind. If I power cycle the RX then they connect immediately (so implies the TX is correctly re-attempting).

## Steps to Reproduce

1. Have both devices in airport mode, 60s wifi timeout

2. Start RX, wait 61s and wifi is activated on the RX.

3. Start TX. It will fail to connect.

4. Power cycle RX. TX will connect instantly.

5. [extended] While in the failure mode at step 3, press Bind initiation button and it goes a bit weird -- the bind message stays on the TX screen, and the bind message comes out on the TX USB (I had to decode it because it wasn't mavlink). Worked fine before Airport mode was enabled, don't know if it is sending over the radio as well but appearing on the USB side was odd.

## Possible Solution (Not obligatory)

I've been looking at the code and it's non-obvious but clearly must be related to some startup state setting because it's totally consistent if done in the order described, and consistently working in the other orders too.

## Details

It worked perfectly before Airport mode was enabled so that's related. The weird bind button behaviour may or may not be related. It also works if the TX comes online before the RX goes into wifi (I think that's the time limit involved)

Tell me how I can get more details?

## Your Environment

- TX hardware: RadioMaster Bandit 900MHz connected over USB to a macbook
- RX hardware: RadioMaster BR1 900MHz
- Handset model: None
- OpenTX version (including nightly number): n/a because no...handset?
- ExpressLRS version (TX & RX MUST MATCH): 3.5.6
- Packet Rate: 200Hz
- Telemetry Ratio: 1:2
- user_defines:

pkendall64, 2025-08-09:
This is 100% expected behaviour.
There is a timer, default 60s on the RX. If there is no packets received from a TX in that time, the radio side is shut down and the wifi is started.

Bwooce, 2025-08-09:
Thanks! 

1. do you want a PR for some more internal docs on "how the RX behaves over time" aka "lifecycle of an RX"?
1. Is my understanding is correct that If I connect once within that 60s and then disconnect then the radio stays active forever as well as the WIFI being enabled after 60s? Or the wifi never enables if a TX was ever connected successfully, and I've just been seeing a phantom SSID echo on my laptop? (probable, which would explain my connection difficulty...). 
1. The reasoning/philosophy behind this is to not have the thousand radios on the tarmac trying to bind to the radio you just powered on (unbound)? or power savings for situations where eLRS won't ever be active (but...wifi isn't free)? If the RX is bound already then a small duty-cycle re-energising of the LoRa side might be a nice feature (for...me) if that's a viable path for me to poke at, on for .5s every 30s or something? Chesterton's fence etc, I'm a bit off the beaten path with my "normally detached rover" use case.

mha1, 2025-08-09:
Just set auto WiFi to 0s to disable this feature.

pkendall64, 2025-08-09:
@Bwooce 
w.r.t to yor points.

1. Always happy to receive documentation PRs
2. 100% correct. If you connect with a TX it means you have had control, so we keep the radio active and looking for signals because you may have had a failsafe, by going out-of-range or some-such.
3. The reasoning is that if you you really should have your transmitter turned on first. If you don't connect in 60 seconds, then you probably want to connect to the the wifi to configure the receiver.

Bwooce, 2025-08-09:
I'll write some docs and ponder the rest, apologies for the WAI bug report and thanks for the correction.

closed 2025-08-09 by Bwooce